wiedehopf/adsb-scripts #4: Problem with pi4 adsb-script

lordkane, 2020-10-10:
hi i have a problem with install readsb-install.sh
i have a error
dpkg-checkbuilddeps: Fehler: Nicht erfüllte Bauabhängigkeiten: librtlsdr-dev

wiedehopf, 2020-10-10:
Install the package manually, see if that works.

```
apt-get update
apt-get install --no-install-recommends --no-install-suggests -y git build-essential debhelper libusb-1.0-0-dev \
    librtlsdr-dev librtlsdr0 pkg-config dh-systemd \
    libncurses5-dev lighttpd zlib1g-dev zlib1g
```

lordkane, 2020-10-10:
Does not workGesendet mit der GMX Mail AppAm 10.10.20 um 19:38 schrieb wiedehopf
                
            
            
                
                    Von: "wiedehopf" <notifications@github.com>Datum: 10. Oktober 2020An: "wiedehopf/adsb-scripts" <adsb-scripts@noreply.github.com>Cc: "Author" <author@noreply.github.com>,"lordkane" <bonzai24@gmx.de>Betreff: Re: [wiedehopf/adsb-scripts] Problem with pi4 adsb-script (#4)
                
                
Install the package manually, see if that works.
apt-get update
apt-get install --no-install-recommends --no-install-suggests -y git build-essential debhelper libusb-1.0-0-dev \
    librtlsdr-dev librtlsdr0 pkg-config dh-systemd \
    libncurses5-dev lighttpd zlib1g-dev zlib1g

[
{
"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/wiedehopf/adsb-scripts/issues/4#issuecomment-706585633",
"url": "https://github.com/wiedehopf/adsb-scripts/issues/4#issuecomment-706585633",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

lordkane, 2020-10-10:
Die folgenden Pakete haben unerfüllte Abhängigkeiten:
 librtlsdr-dev : Hängt ab von: librtlsdr0 (= 0.6-1+rpt1) aber 0.6git soll installiert werden
E: Probleme können nicht korrigiert werden, Sie haben zurückgehaltene defekte Pakete.

wiedehopf, 2020-10-10:
you'll have to remove the blocking packet.

sudo apt remove librtlsdr0

then try again to install.

closed 2020-10-10 by wiedehopf