wiedehopf/adsb-scripts #5: unknown option or argument --build-profiles=rtlsdr (and --no-sign)

larsen0815, 2020-10-28:
I tried to install "readsb-install.sh" and got this error message as those long params don't exist in Debian Jessie (Raspbian).

````
dpkg-buildpackage: unknown option or argument --build-profiles=rtlsdr
dpkg-buildpackage: unknown option or argument --no-sign
````

Solution is to use the short form:
````if ! dpkg-buildpackage -b -Prtlsdr -uc````

wiedehopf, 2020-10-28:
The solution is not to use Jessie?

Anyhow .. working around the fossil and pushing a commit in a moment :)
Just for the record, you need three options to get --no-sign:
```
-if ! dpkg-buildpackage -b --build-profiles=rtlsdr --no-sign
+if ! dpkg-buildpackage -b -Prtlsdr -ui -uc -us
```

closed 2020-10-28 by wiedehopf

larsen0815, 2020-10-28:
That would be the other solution, yes =)
Have installed the Pi as a FR24 box in 2016. Never change a running system (which I did by installing the adsbexchange stuff *g*).

Compilation was fine, afaict. Any possible downsides of not having used "-us"?

wiedehopf, 2020-10-28:
If it works, it works.

Just saying it's not equivalent to --no-sign.