wiedehopf/adsb-scripts #8: readsb install fails

imesg, 2021-01-23:
How do I fix this?

Thanks for making this available.

Can supply more info is needed.

Cloning into '/usr/local/share/adsb-wiki/readsb-install/git'...
remote: Enumerating objects: 109, done.
remote: Counting objects: 100% (109/109), done.
remote: Compressing objects: 100% (94/94), done.
remote: Total 109 (delta 20), reused 57 (delta 8), pack-reused 0
Receiving objects: 100% (109/109), 272.34 KiB | 297.00 KiB/s, done.
Resolving deltas: 100% (20/20), done.
Checking connectivity... done.
dpkg-buildpackage: unknown option or argument -ui

imesg, 2021-01-23:
Sorry here's a little more:

Cloning into '/usr/local/share/adsb-wiki/readsb-install/git'...
remote: Enumerating objects: 109, done.
remote: Counting objects: 100% (109/109), done.
remote: Compressing objects: 100% (94/94), done.
remote: Total 109 (delta 20), reused 57 (delta 8), pack-reused 0
Receiving objects: 100% (109/109), 272.34 KiB | 297.00 KiB/s, done.
Resolving deltas: 100% (20/20), done.
Checking connectivity... done.
dpkg-buildpackage: unknown option or argument -ui

Use --help for program usage information.
Something went wrong building the debian package, exiting!

wiedehopf, 2021-01-23:
What system are you on?
```
cat /etc/os-release
```

I'd suspect you're just on a very old system.
https://github.com/wiedehopf/adsb-wiki/wiki/Building-readsb-from-source

This might help you as you can modify commands if they don't work ...

imesg, 2021-01-24:
Thanks. You were exactly correct. R pi 4 had raspbian 'jessie' installed. I upgraded to 'stretch' and then upgraded to 'buster'. Then installed readsb.

After wrangling with lighttpd a bit, it works. Very nice.

Thanks

closed 2021-05-01 by wiedehopf