wiedehopf/adsb-scripts #11: Add a description to the repository

mendel5, 2021-05-23:
Hi, can you add a description to the repository? How about this one:

> Solutions to common problems using dump1090 variants and ADS-B feeders.

Thanks!

wiedehopf, 2021-05-23:
What does it matter?

closed 2021-05-23 by wiedehopf

mendel5, 2021-05-23:
Thanks for adding it.

Having a description improves the [discoverability](https://en.wikipedia.org/wiki/Discoverability) of a repository. If, for example, an interested user found this repo on his [https://github.com/explore](https://github.com/explore) page, a useful description helps him to identify interesting content. By improving the discoverability, more people might interact with this repo and thereby improve it for everyone else:
> Given enough eyeballs, all bugs are shallow - Linus' Law