wiedehopf/adsb-scripts #12: question: autogain1090- disable RTL AGC ( -10)

messschuh, 2021-07-12:
as far as i know "-10" activate the AGC of RTLchip which isnt true AGC. is that right?

i want disable the "-10" setting, i tried it by removing  "-10" from /etc/default/autogain1090 (using readsb not dump1090-fa).
( stock  ga=(.... 48.0 49.6 -10) changed to> ga=(....48.0 49.6) 
but autogain1090 set "-10" anyway. isnt it the right way or is it not possible to disable it ?
of course i restart the service and/or reboot the system.

wiedehopf, 2021-07-12:
If you want a fixed gain setting, don't use autogain.

wiedehopf, 2021-07-12:
Also -10 is equivalent to a gain of 56 or so in regards to ADS-B due to the short messages, AGC never reacts.
https://github.com/wiedehopf/adsb-wiki/wiki/Optimizing-gain

closed 2021-07-12 by wiedehopf

messschuh, 2021-07-12:
thanks for fast response.
no, i dont want a fix gain setting, i want use autogain1090 but i dont want autogain1090 set the gain to "-10"! is that possible?

wiedehopf, 2021-07-12:
either autogain switches back and forth between 49.6 and -10 .. then you just set a fixed gain of 49.6
otherwise it won't go to -10 ....

anyhow .... run the install again, i changed it so the ga= works in the default file.
what you're doing doesn't make much sense but whatever ....

messschuh, 2021-07-12:
thx, i will try it. 
why should it not make sense? most time planes i receive are far away, so signal is low (and gain high) but sometimes few planes are very close with very! strong signal and i dont want they "overload" the receiver input. i´m not really familar how it really works, if u have some informations sources (links) about it...?

wiedehopf, 2021-07-13:
https://github.com/wiedehopf/adsb-wiki/wiki/Optimizing-gain

The autogain will only run once at night.
It's not dynamic gain adjustment, it just finds a good compromise.

-10 can be a valid outcome of the logic so you shouldn't remove it.
well if it toggles between 49.6 and -10 you should remove it.
but then you can just as well turn off autogain and just set 49.6

sorry i don't want to explain this further, good luck understanding it on your own :)
once you understand it, you won't need autogain.