wiedehopf/adsb-scripts #13: upcoming changes to /etc/default/dump1090-fa

mutability, 2021-07-26:
Hi, wanted to give you a heads-up that I've got some changes planned to how dump1090-fa's config file works in 6.0, which are likely to break your scripts that do sed-driven replacements on that file.

The current changes are on the dev branch: https://github.com/flightaware/dump1090/blob/dev/debian/dump1090-fa.default

The tl;dr is that rather than having the config directly include a long list of command-line options, instead the config has separate vars for different features and the start-dump1090-fa script assembles those into options.

If there's anything unworkable in the new format (or if there are settings you want to see in there that aren't directly supported), let me know and I'll see what I can do?

wiedehopf, 2021-07-26:
Oh the old style configuration file.
I'll figure it out.

You could include some keywords in the new configuration file to indicate it's the new style.
That way you could without much hassle even retain compatibility to the old style.
That would also save you from many questions when people don't update their configuration file when updating dump1090-fa.
Yes i know it's their own fault but ... still doesn't reduce the amount of forum activity it produces.

CONFIG_STYLE=XYZ
something like that?

How close is 6.0, a month or less?

mutability, 2021-07-26:
Yeah, like d1090-mutability but no debconf (debconf is such a pain)

I'll see what I can do with something like CONFIG_STYLE. I do auto-upgrade existing config files to the new style on package upgrade so hopefully existing installs are OK either way.

6.0 is maybe a month away, yeah.

wiedehopf, 2021-07-26:
The auto-upgrade depends on people hitting new during the update process?
Or do you force it?

Anyway not an issue i'll change the stuff that messes with it.
Should only be autogain and airspy-conf really.

mutability, 2021-07-26:
In postinst if the config file is still old-style then I read it and regenerate a new config based on the old options. see https://github.com/flightaware/dump1090/blob/dev/debian/upgrade-config

mutability, 2021-07-26:
You can look for `CONFIG_STYLE=6`: https://github.com/flightaware/dump1090/commit/899ee2530cced9ba86867a80b49f65213d1073ef#diff-7c16bb3e4fbf498e4d5c06ee25690373efefa0e596edce64f8384ccfb5136319R62

closed 2024-04-08 by wiedehopf