wiedehopf/adsb-scripts #14: Update failed (Initialization of state variables in list context currently forbidden)

giddyhup, 2021-08-10:
I wanted to update readsb via the update/install script and the following error happened:

```
dpkg-buildpackage: info: host architecture armhf
 dpkg-source --before-build git
 debian/rules clean
dh clean --with=systemd
Initialization of state variables in list context currently forbidden at /usr/share/perl5/Debian/Debhelper/Dh_Lib.pm line 2032, near ");"
BEGIN not safe after errors--compilation aborted at /usr/share/perl5/Debian/Debhelper/Dh_Lib.pm line 2716.
Compilation failed in require at /usr/bin/dh line 15.
BEGIN failed--compilation aborted at /usr/bin/dh line 15.
debian/rules:54: recipe for target 'clean' failed
make: *** [clean] Error 2
dpkg-buildpackage: error: debian/rules clean gave error exit status 2
Something went wrong building the debian package, exiting!

```

This is under buster Raspberry OS.

giddyhup, 2021-08-10:
Should I update perl?

```
 perl -v

This is perl 5, version 24, subversion 1 (v5.24.1) built for arm-linux-gnueabihf-thread-multi-64int
(with 86 registered patches, see perl -V for more detail)

```

wiedehopf, 2021-08-10:
Just update the pi?
```
sudo apt update && sudo apt dist-upgrade
```

Can't reproduce the error.

giddyhup, 2021-08-10:
Thanks for the quick feedback. You were right, the pi was in dire need of an update. A simple update/upgrade wasn't enough, a dist-upgrade was needed. Writing the web server conf file failed but since this was an upgrade and not a fresh install it didn't matter. I restarted the service and everything looks good.

(The conf file cp error:
```
Processing triggers for man-db (2.8.5-2) ...
Package installed!
cp: cannot stat 'debian/lighttpd/*': No such file or directory
[ERROR] Error in line 82 when executing: cp -n debian/lighttpd/* /etc/lighttpd/conf-available
```
)

closed 2021-08-10 by giddyhup

wiedehopf, 2021-08-10:
Sorry that was me messing with my readsb repository.
Should have fixed it rather quickly though.

wiedehopf, 2021-08-10:
So feel free to rerun, it should all work now.

giddyhup, 2021-08-10:
Thanks for the explanation. If I am not mistaken, none of the steps following the failure are needed in my situation so I guess I'll leave it the way it is.

wiedehopf, 2021-08-10:
Did it update tar1090 as well? And restart readsb?
You can always do that manually i suppose.

giddyhup, 2021-08-10:
I actually updated tar1090 on its own before readsb and also restarted the service manually.

wiedehopf, 2021-08-10:
Btw i just fixed the graphs1090 install script ... should really test that stuff :)
https://github.com/wiedehopf/graphs1090#graphs1090

giddyhup, 2021-08-10:
Ok, ok. I messed with my gain settings and used the occasion while fixing it to run the full install/update script. Everything worked nicely.

giddyhup, 2021-08-10:
The range diagram is new to me, neat.