wiedehopf/adsb-scripts #16: Install not working on Debian GNU/Linux 11 (bullseye)

binaryanomaly, 2021-09-05:
Building it fails on Debian GNU/Linux 11 (bullseye). Any idea why?

`cc -Wdate-time -D_FORTIFY_SOURCE=2 -DMODES_READSB_VERSION=\""wiedehopf git: 98e86ab (commited: Fri Sep 3 15:25:38 2021 0200)"\" -D_GNU_SOURCE -D_FORTIFY_SOURCE=2 -fstack-protector-strong -Wformat -Werror=format-security -DAIRCRAFT_HASH_BITS=15 -DENABLE_RTLSDR -std=c11 -g -W -D_DEFAULT_SOURCE -Wall -Werror -fno-common -O2 -g -O2 -ffile-prefix-map=/usr/local/share/adsb-wiki/readsb-install/git=. -fstack-protector-strong -Wformat -Werror=format-security -march=native -Wno-format-truncation -c receiver.c -o receiver.o
globe_index.c: In function ‘load_blob’:
globe_index.c:1293:116: error: array subscript 8261 is above array bounds of ‘struct craftArray[8261]’ [-Werror=array-bounds]
 1293 | (stderr, "hex: %06x, old_index: %d, new_index: %d, GLOBE_MAX_INDEX: %d\n", a->addr, Modes.globeLists[new_index].len, new_index, GLOBE_MAX_INDEX );
      |                                                                                     ~~~~~~~~~~~~~~~~^~~~~~~~~~~

In file included from globe_index.c:1:
readsb.h:439:23: note: while referencing ‘globeLists’
  439 |     struct craftArray globeLists[GLOBE_MAX_INDEX+1];
      |                       ^~~~~~~~~~
readsb.h:639:22: note: defined here ‘Modes’
  639 | extern struct _Modes Modes;
      |                      ^~~~~
cc -Wdate-time -D_FORTIFY_SOURCE=2 -DMODES_READSB_VERSION=\""wiedehopf git: 98e86ab (commited: Fri Sep 3 15:25:38 2021 0200)"\" -D_GNU_SOURCE -D_FORTIFY_SOURCE=2 -fstack-protector-strong -Wformat -Werror=format-security -DAIRCRAFT_HASH_BITS=15 -DENABLE_RTLSDR -std=c11 -g -W -D_DEFAULT_SOURCE -Wall -Werror -fno-common -O2 -g -O2 -ffile-prefix-map=/usr/local/share/adsb-wiki/readsb-install/git=. -fstack-protector-strong -Wformat -Werror=format-security -march=native -Wno-format-truncation -c aircraft.c -o aircraft.o
cc1: all warnings being treated as errors
make[2]: *** [Makefile:94: globe_index.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[2]: Leaving directory '/usr/local/share/adsb-wiki/readsb-install/git'
make[1]: *** [debian/rules:46: override_dh_auto_build] Error 2
make[1]: Leaving directory '/usr/local/share/adsb-wiki/readsb-install/git'
make: *** [debian/rules:61: build] Error 2
dpkg-buildpackage: error: debian/rules build subprocess returned exit status 2
Something went wrong building the debian package, exiting!`

wiedehopf, 2021-09-05:
Try again with the updated source.

binaryanomaly, 2021-09-05:
Thanks, works now 👌🏻

closed 2021-09-05 by wiedehopf