wiedehopf/adsb-scripts #17: awk syntax error

upD8R, 2022-01-21:
Hi there,

just setting up a new RPi with your very cool scripts but after installing autogain1090 U get an awk syntax error the "basic" stuff work as expected as far as I can tell):
`pi@flight:~ $ sudo autogain1090
awk: line 1: syntax error at or near }
awk: line 1: syntax error at or near }
No gain change needed, percentage of messages >-3dB is in nominal range. (0,968%)`

Any idea?

wiedehopf, 2022-01-21:
does it persist or was it only happening on first run?

wiedehopf, 2022-01-21:
Run the install again, see if the error is gone.

upD8R, 2022-01-21:
It persists, also after 2nd installation.

Looks like a variable is not (yet) populated? But I don't know, which awk is throwing this error.

wiedehopf, 2022-01-21:
Edit /usr/local/bin/autogain1090 and add a set -x at the start of the script.

upD8R, 2022-01-21:
Something "," vs "."?
```
+ awk 'BEGIN{ exit (0,000 > 0.5) }'
awk: line 1: syntax error at or near }
+ awk 'BEGIN{ exit (0,000 < 7.0) }'
awk: line 1: syntax error at or near }

```

wiedehopf, 2022-01-21:
Yeah it's your locale .... i really don't know why awk depends on locale it's very annoying.

Ok ... try another install, should  be fixed.

upD8R, 2022-01-21:
Many thanks, by Googling I was also ending up at my locale ;-)

It is fixed now, thanks indeed for your outstanding support!

closed 2022-01-21 by wiedehopf