wiedehopf/adsb-scripts #19: stats.json not found?

nar1117, 2022-02-02:
Alright, so I have dump1090-mutability installed, sending data to FR24feed and flightaware. Just found your script for the gain through the flightaware forums. Installed it, and I'm now running the 1-hour long For-loop to get an initial gain setting. It's giving the error: "stats.json not found, is the decoder running?"

Watching my adsb map at the same time, I'm pretty sure it's still changing the gain, but the command "sudo journalctl -eu autogain1090" shows that nothing is being written to the logs.

Will this work at all for a standard dump1090-mutability installation? I'm considering just reinstalling dump1090 with your scripts, but if I can keep what I have already and use the gain script that would be ideal!

wiedehopf, 2022-02-02:
That autogain doesn't say that it works for dump1090-mutability :)
So no it won't work for that.

Just install readsb, you won't regret it.
 https://github.com/wiedehopf/adsb-wiki/wiki/Raspbian-Lite:-ADS-B-receiver

nar1117, 2022-02-02:
Whelp, I just went ahead and modified the script a bit! changed the $APP variables to point to dump1090-mutability.

But yes! I think I'll still install readsb. Thanks!

wiedehopf, 2022-02-02:
The configuration file is different, you'd have to change the code that changes the gain.

nar1117, 2022-02-02:
Ah, yep I see that now. All good! Thanks for your help.

closed 2022-02-02 by wiedehopf