wiedehopf/adsb-scripts #20: 403 errors when accessing web interface

beowulf222, 2022-02-06:
Hi,

I used the install.sh script for the Bundle install for dump1090fa. So far so good.

However, when I try to access the three different web interfaces, only /dump1090-fa is accessible. Both /dump1090 and /tar1090 return a 403 Error in the browser. 
My guess is it's a permission error but I cannot find anything in the lighthttp log files.

Does anyone here have any pointers as to where to adjust permissions or find out more?

-- nick

wiedehopf, 2022-02-06:
Try the tar1090 install separately and let me know if there are any errors:
 https://github.com/wiedehopf/tar1090#tar1090

beowulf222, 2022-02-06:
Installation via the script works but the process is not running

```
● tar1090.service - tar1090 - compress dump1090 json data
   Loaded: loaded (/lib/systemd/system/tar1090.service; enabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Sun 2022-02-06 17:50:08 +08; 1s ago
  Process: 30780 ExecStart=/usr/local/share/tar1090/tar1090.sh /run/tar1090 /run/dump1090-fa (code=exited, status=203/EXEC)
 Main PID: 30780 (code=exited, status=203/EXEC)
```

wiedehopf, 2022-02-06:
Show the log please .... not the status.
```
sudo journalctl -u tar1090 | tail -n15
```

But really this is unrelated to the 403, the webinterface should work with reduced functionality without this service.

beowulf222, 2022-02-06:
```
Feb 06 18:33:30 raspi systemd[1]: tar1090.service: Scheduled restart job, restart counter is at 12425.
Feb 06 18:33:30 raspi systemd[1]: Stopped tar1090 - compress dump1090 json data.
Feb 06 18:33:30 raspi systemd[1]: Started tar1090 - compress dump1090 json data.
Feb 06 18:33:30 raspi systemd[6204]: tar1090.service: Failed to execute command: Permission denied
Feb 06 18:33:30 raspi systemd[6204]: tar1090.service: Failed at step EXEC spawning /usr/local/share/tar1090/tar1090.sh: Permission denied
Feb 06 18:33:30 raspi systemd[1]: tar1090.service: Main process exited, code=exited, status=203/EXEC
Feb 06 18:33:30 raspi systemd[1]: tar1090.service: Failed with result 'exit-code'.
Feb 06 18:34:00 raspi systemd[1]: tar1090.service: Service RestartSec=30s expired, scheduling restart.
Feb 06 18:34:00 raspi systemd[1]: tar1090.service: Scheduled restart job, restart counter is at 12426.
Feb 06 18:34:00 raspi systemd[1]: Stopped tar1090 - compress dump1090 json data.
Feb 06 18:34:00 raspi systemd[1]: Started tar1090 - compress dump1090 json data.
Feb 06 18:34:00 raspi systemd[6379]: tar1090.service: Failed to execute command: Permission denied
Feb 06 18:34:00 raspi systemd[6379]: tar1090.service: Failed at step EXEC spawning /usr/local/share/tar1090/tar1090.sh: Permission denied
Feb 06 18:34:00 raspi systemd[1]: tar1090.service: Main process exited, code=exited, status=203/EXEC
Feb 06 18:34:00 raspi systemd[1]: tar1090.service: Failed with result 'exit-code'.
```

wiedehopf, 2022-02-06:
Combined with the 403 ..... i'd say your sd-card is broken.

This error can't happen if the sd-card is working properly.

beowulf222, 2022-02-06:
Well, I have been using the same SD card for 3+ years, and the PI has been running non-stop (occasionally exiting not gracefully), but I don't observe any other issues with the SD card.

wiedehopf, 2022-02-06:
My install script for tar1090 is used by so many people and if i see two unrelated errors, there's not something wrong with the scripts but rather with the pi.

Try the uinstall, then try again with the install
 https://github.com/wiedehopf/tar1090#remove--uninstall

wiedehopf, 2022-02-06:
If the uninstall / install helps .... it's very likely the card as the install hasn't changed.
See this error says the execution permissions are not set on the file.
If you install with my install script, the file will have the correct permissions.
If it doesn't ... there's something wrong and not with my script.

perxmys, 2023-05-19:
This may not be the cause of the problem that the original poster encountered, but have found a condition that will lead to the eventual 403 forbidden error when trying to use the web interface.

If the install script is run while a umask is set to something different other than the raspbian default of 0022, then the permissions for the resulting binaries/directories/etc. might not be set correctly.

I noticed this after doing two installations on separate devices, but ran into the 403 error on the second device, and then subsequently found the permissions were off. Was baffled by why it the install worked on the first device, but not the second, until I realized that I had run the install script on the second device after copying over a .bashrc that had some of the usual customizations over, and which included a 'umask 0077' line in it, leading to the incorrect permissions being set for the resulting files (whereas on the first device I had run the install script before copying the .bashrc over).

wiedehopf, 2023-05-19:
I suppose i could explicitely overwrite perms for all files being installed ..

perxmys, 2023-05-19:
That could be helpful. I don't know how common you'd run into this though.

Perhaps a warning (or a check) to make sure the umask is 022 before compiling would work, at least for now.

closed 2024-04-08 by wiedehopf