wiedehopf/adsb-scripts #21: Unable to work with HackRF one

chekmeny, 2022-02-20:
Hi,
  I just changed my decoder from ‘rtlsdr’ to ‘hackrf one’, but I found that the dump1090 could not work anymore, which shows ‘ no supported device found ‘. I’ve tried to change the ‘RECEIVER = rtlsdr’ to ‘RECEIVER = hackrf’ in the dump1090 config file, but it still not work. I don’t know if the version from the automated install is not supported to hackrf? Or is there any way to solve this problem?
  Thanks!

wiedehopf, 2022-02-20:
Don't use a hackrf for ADS-B ... 
And if you're talking about dump1090-fa ... you're opening an issue on the wrong repository.

closed 2022-02-20 by wiedehopf

geerlingguy, 2025-07-17:
I saw the file at `/etc/default/readsb`, and it had the `rtlsdr` defined. I tried just changing that after re-running the modified install script, but it's saying:

```
Jul 16 22:47:30 pi-sdr readsb[618508]: SDR type 'hackrf' not recognized; supported SDR types are:
Jul 16 22:47:30 pi-sdr readsb[618508]:   rtlsdr
Jul 16 22:47:30 pi-sdr readsb[618508]:   modesbeast
Jul 16 22:47:30 pi-sdr readsb[618508]:   gnshulc
Jul 16 22:47:30 pi-sdr readsb[618508]:   ifile
Jul 16 22:47:30 pi-sdr readsb[618508]:   none
Jul 16 22:47:30 pi-sdr readsb[618508]: ERROR: Unknown device type:hackrf
Jul 16 22:47:30 pi-sdr readsb[618508]: Error parsing the given command line parameters, check readsb --usage and readsb --help for valid parameters.
```

geerlingguy, 2025-07-17:
Ah... found `RTLSDR=yes` elsewhere in the install file. I'm adding in `HACKRF=yes` in those same places (repeated 4x) and re-running...

```
sudo bash readsb-install.sh
```

I also had to modify the `RECEIVER_OPTIONS` to remove the rtlsdr-specific options `--gain` and `--ppm`.

But then, after running the install again, I got:

```
Jul 16 22:59:51 pi-sdr readsb[623845]: Opening HackRF: 0
Jul 16 22:59:51 pi-sdr readsb[623845]: Failed to open hackRF: HackRF not found
```

Here's the info on my HackRF One:

```
$ hackrf_info 
hackrf_info version: 2022.09.1
libhackrf version: 2022.09.1 (0.7)
Found HackRF
Index: 0
Serial number: 0000000000000000436c63dc38593463
Board ID Number: 2 (HackRF One)
Firmware Version: 2024.02.1 (API:1.08)
Part ID Number: 0xa000cb3c 0x006f435e
Hardware Revision: unknown
Hardware appears to have been manufactured by Great Scott Gadgets.
Hardware supported by installed firmware:
    HackRF One
```

And _FINALLY_, I can get at it by passing the serial number directly in the `RECEIVER_OPTIONS`, so the resulting `` file reads:

```
RECEIVER_OPTIONS="--device 0 --device-type hackrf --device 0000000000000000436c63dc38593463"
DECODER_OPTIONS="--max-range 450 --write-json-every 1"
NET_OPTIONS="--net --net-heartbeat 60 --net-ro-size 1250 --net-ro-interval 0.05 --net-ri-port 30001>
JSON_OPTIONS="--json-location-accuracy 2 --range-outline-hours 24"
```

wiedehopf, 2025-07-17:
That part about /etc/default/readsb in the install script is conditional on a variable.
Usually the default file is taken from the readsb repo in the debian directory and not overwritten once it is present.

Glad you got it working, you have some nice videos.

IIRC the hackrf is not quite sensitive enough for good ADS-B reception.
If you have an LNA handy you can possibly improve it.

The code supporting the hackrf uses the same limited sample rate of 2.4 MSPS and i'm not sure the hackrf would perform better with higher sample rates but it's possible.

Specialized rtl-sdr devices are probably the cheapest way to have a permanent ADS-B receiver that's working well, some hardware is listed here: https://github.com/wiedehopf/adsb-wiki/wiki/adsb-receiver-shopping-list
One of the best possible setups would be a 1090 filtered LNA with an airspy mini.
If you're looking for an all in one pi image for ADS-B, take a look at http://adsb.im/
(i don't think it supports hackrf though)

You landed on this page? https://github.com/wiedehopf/adsb-scripts/wiki/Automatic-installation-for-readsb

I guess this is the 2nd person trying to use an hackrf so i should possibly add notes.
Even if it's not really well supported / reception isn't that good.