wiedehopf/adsb-scripts #23: Add /usr/local/bin/ to path

Somebodyisnobody, 2022-07-22:
`readsb-gain` and `readsb-set-location` are not accessible like described here:
https://github.com/wiedehopf/adsb-scripts/blob/3028c5bd1cc142a11450c02b2ac787842c3d0a88/readsb-install.sh#L248-L251
This is because /usr/local/bin/ is not added to the path variable. I suggest to rectify this OS-misconfiguration with the install script.

wiedehopf, 2022-07-22:
What OS is that?

I won't F with PATH .... i'd rather go back to display /usr/local/bin

Somebodyisnobody, 2022-07-22:
I have an Proxmox LXC-container with Debian 11 (debian-11-standard_11.3-1_amd64.tar.zst). It's a clean installation.

wiedehopf, 2022-07-22:
yeah i'll display the full path again maybe

not changing PATH variables that's messy.

Somebodyisnobody, 2022-07-22:
or as hint like "the script is located in /usr/local/bin/"

closed 2022-07-22 by wiedehopf