wiedehopf/adsb-scripts #26: Outdated version FR24 feeder

jdschuitemaker, 2023-01-19:
Today I received an email from FR24 that my feeder is not up to date (version 1.0.29-8) and that I have to update to the latest version (1.0.34-0) before Feb 28, 2023. I have been following your instruction to update in section 10 here [https://github.com/wiedehopf/adsb-wiki/wiki/Raspbian-Lite:-ADS-B-receiver](https://github.com/wiedehopf/adsb-wiki/wiki/Raspbian-Lite:-ADS-B-receiver) but it seems I am still on an older version (frfe24feed output is showing version  1.0.30-3/generic).

The instruction in the email from FR24 is to run the following commands - one at a time:
```
sudo chmod +x /usr/lib/fr24/fr24feed_updater.sh
sudo /usr/lib/fr24/fr24feed_updater.sh
```

But the folder fr24 does not exist in lib. Can you help please?

Thank you!

wiedehopf, 2023-01-19:
Their updater breaks stuff.

They will have to live with the outdated version for a bit.
(until i update the script)

If you follow their install instructions, be ready for updater breaking stuff in the future at your own peril.

closed 2023-01-19 by wiedehopf

Falconcrest, 2023-01-19:
...and I thought, once again just my problem!
I agree, the same thing happened to me.
But I was too hasty and updated fr24feed and thereby destroyed the installation. fr24feed did not start anymore. Then I remembered these pages here and found the description and solution of the problem.

Under adsb/wiki, point 10 Feed FR24 (optional) tells you what to do.

`sudo bash -c "$(wget -O - https://github.com/wiedehopf/adsb-scripts/raw/master/fr24-nopackage.sh)"` 

Since then everything has been working fine again.

Version 1.0.26-9 was installed before the email from Flightradar24. After the unsuccessful update it was version 1.0.34-0, after the "repair" it is version 1.0.30-3. This is not the latest, but as written above, the script will probably be updated.

Thanks in advance for that!