wiedehopf/adsb-scripts #28: FR24 updater on Wiki page

mobisat, 2023-01-22:
I realise this may not be in the correct section as it relates to the Wiki page.
[https://github.com/wiedehopf/adsb-wiki/wiki/Raspbian-Lite:-ADS-B-receiver](url)

I received an e-mail from Flightradar that the current version that I was running was soon to be phased out.

"We have noticed that you are using an older version of Flightradar24 data sharing software (1.0.29-8) that we are phasing out. We would therefore like you to upgrade to the latest version (1.0.34-0) before Feb 28, 2023."

I went to the wiki page which I used to install it originally using the command sudo bash -c "$(wget -O - https://github.com/wiedehopf/adsb-scripts/raw/master/fr24-nopackage.sh)"

It did update my setup however, only to Software Version 1.0.30-3

Would it be possible to update your script to allow it to upgrade to the latest version 1.0.34-0?

wiedehopf, 2023-01-22:
someone already put a PR....
it's changed since yesterday: https://raw.githubusercontent.com/wiedehopf/adsb-scripts/master/fr24-nopackage.sh

closed 2023-01-22 by wiedehopf

Falconcrest, 2023-01-24:
> it's changed since yesterday: https://raw.githubusercontent.com/wiedehopf/adsb-scripts/master/fr24-nopackage.sh

Thanks very much!