wiedehopf/adsb-scripts #30: FR24 feeder install/update error

motu876, 2023-02-05:
Hi,

if I try to update my FR24 feeder with the fr24-nopackage script terminates withe a permission denied error (see error log below). Is this a server error or problem with the script?

-- oliver

```
sudo bash -c "$(wget -O - https://github.com/wiedehopf/adsb-scripts/raw/master/fr24-nopackage.sh)"
--2023-02-05 14:29:34--  https://github.com/wiedehopf/adsb-scripts/raw/master/fr24-nopackage.sh
Resolving github.com (github.com)... 140.82.121.4
Connecting to github.com (github.com)|140.82.121.4|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://raw.githubusercontent.com/wiedehopf/adsb-scripts/master/fr24-nopackage.sh [following]
--2023-02-05 14:29:34--  https://raw.githubusercontent.com/wiedehopf/adsb-scripts/master/fr24-nopackage.sh
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 185.199.110.133, 185.199.108.133, 185.199.111.133, ...
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|185.199.110.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 1004 [text/plain]
Saving to: ‘STDOUT’

-                             100%[==============================================>]    1004  --.-KB/s    in 0s

2023-02-05 14:29:35 (14.7 MB/s) - written to stdout [1004/1004]

Hit:1 http://archive.raspberrypi.org/debian bullseye InRelease
Hit:2 http://raspbian.raspberrypi.org/raspbian bullseye InRelease
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
All packages are up to date.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
perl is already the newest version (5.32.1-4+deb11u2).
wget is already the newest version (1.21-1+deb11u1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package fr24feed
fr24.deb: Permission denied
```

wiedehopf, 2023-02-05:
problem with your box i'd guess.

motu876, 2023-02-07:
It works now again. Not sure what was wrong. After several try’s and reboot’s it works…

closed 2023-02-07 by motu876