wiedehopf/adsb-scripts #31: Question related to the autogain settings

matrixxus, 2023-02-21:
Hello, and sorry for posting this in the issues section, as it's not an error but a question.

It is about: https://github.com/wiedehopf/adsb-scripts/wiki/Automatic-gain-optimization-for-readsb-and-dump1090-fa
Thanks for publishing it, works great.
My only question is: running your script on a PiAware installation, should I disable the Piaware built-in autogain setting?

```
RECEIVER_GAIN=43.9
ADAPTIVE_DYNAMIC_RANGE=yes
ADAPTIVE_DYNAMIC_RANGE_TARGET=
ADAPTIVE_BURST=no
ADAPTIVE_MIN_GAIN=
ADAPTIVE_MAX_GAIN=
```
from the file:  /etc/default/dump1090-fa
Wondering if this should be changed somehow.

matrixxus, 2023-02-22:
Thank you very much for the reply.
I am on the piaware system image. What I was trying to understand was, if when using your 'autogain-install.sh' script, should the built-in piaware auto-gain option be disabled or enabled. Or doesn't matter at all?

wiedehopf, 2023-02-23:
No clue how that interacts, in doubt switch it off.
Either script based autogain or their builtin autogain.

matrixxus, 2023-02-24:
Understood, keeping your autogain only, thank you very much for the explanation.

closed 2023-02-24 by matrixxus