wiedehopf/adsb-scripts #37: readsb fails to build on Ubuntu 24.04 (Noble Nimbat)

abcd567a, 2024-03-21:
Ran following installation script on Ubuntu 24.04 (noble) amd64
`sudo bash -c "$(wget -O - https://github.com/wiedehopf/adsb-scripts/raw/master/readsb-install.sh)"
sudo reboot `

The above script failed to install readsb with following error message:
```
<command-line>: error: "_FORTIFY_SOURCE" redefined [-Werror]
<built-in>: note: this is the location of the previous definition
cc1: all warnings being treated as errors
make: *** [Makefile:136: readsb.o] Error 1
[ERROR] Error in line 115 when executing: make "-j${THREADS}" AIRCRAFT_HASH_BITS=16 RTLSDR=yes OPTIMIZE="$CFLAGS" "$@"

```

abcd567a, 2024-03-21:
Ran the install script again now, still failing with following error message:

```
In file included from readsb.h:101,
                 from readsb.c:55:
threadpool.h:36:14: fatal error: zstd.h: No such file or directory
   36 |     #include <zstd.h>
      |              ^~~~~~~~
compilation terminated.
make: *** [Makefile:136: readsb.o] Error 1
[ERROR] Error in line 115 when executing: make "-j${THREADS}" AIRCRAFT_HASH_BITS=16 RTLSDR=yes OPTIMIZE="$CFLAGS" "$@"
abcd@ubuntu-24:~$
```

wiedehopf, 2024-03-21:
Those installed? The script installs them i'd think.
```
libzstd-dev libzstd1
```

abcd567a, 2024-03-21:
> Those installed? The script installs them i'd think.
> 
> ```
> libzstd-dev libzstd1
> ```

**The  `libzstd-dev` was NOT installed
The `libzstd1` was installed**
```
abcd@ubuntu-24:~$ apt-cache policy libzstd-dev | grep Installed
  Installed: (none)
abcd@ubuntu-24:~$
abcd@ubuntu-24:~$ apt-cache policy libzstd1 | grep Installed
  Installed: 1.5.5+dfsg2-2
```

**Manually installed `libzstd-dev`**
```
abcd@ubuntu-24:~$ sudo apt install libzstd-dev 

abcd@ubuntu-24:~$ apt-cache policy libzstd-dev | grep Installed
  Installed: 1.5.5+dfsg2-2
```

**After manually installing `libzstd-dev`, ran the install script again, and SUCCESS this time**

[image]

wiedehopf, 2024-03-21:
The script installs both packages.
I have no idea why it didn't install for you.

Feel free to remove the package and run the script again ... see if it installs it.

abcd567a, 2024-03-22:
> Feel free to remove the package and run the script again ... see if it installs it.

I tried, and it failed again with error message.

I have also noted that script fails to install librtlsdr0 saying 'not available".

I then checked following part of installation script:
```
 58     packages=(git gcc make libusb-1.0-0-dev librtlsdr-dev librtlsdr0 ncurses-dev ncurses-bin zlib1g-dev zlib1g)
 59     if ! grep -E 'wheezy|jessie' /etc/os-release -qs; then
 60         packages+=(libzstd-dev libzstd1)
 61     fi
```

My conclusion is that **when script fails at installation of `librtlsdr0`, it aborts  installation of remaining packages in the list**. 
Since I have already built and installed dump1090-fa & piaware from source-code, the pacakge librtlsdr-dev, librtlsdr2, as well as other packages (except libzstd-dev) were already installed. As  a result libzstd-dev required to be installed manually. 

I remember that in one of my scripts I had following command: 

`apt install -y package-1 package-2 ...... package-n`

Failure to find / install any one package caused the apt to abort installation of remaining packages. 
I solved this by rewriting script in following manner:
```
apt install -y package-1
apt iinstall -y package-2
..... 
.....
apt install -y package-n
```

[image]

abcd567a, 2024-03-22:
### Package installation command CONTAINING package `librtlsdr0` : </br>Package installation process was aborted
[image]

</br>

### Package installation command WITHOUT package `librtlsdr0` : </br>Package installation process was COMPLETED
[image]

wiedehopf, 2024-03-22:
> I have also noted that script fails to install librtlsdr0 saying 'not available".

But you didn't have that issue before?
Or did you remove the whole apt install part because of the error?

wiedehopf, 2024-04-08:
I've separated the librtlsdr install so it can fail and other packages are still installed.

This really only helps if you already have librtlsdr-dev / librtlsdr0 installed (maybe librtlsdr2 works as well)
I'm letting this settle until it's clear which librtlsdr version will actually be available with Ubuntu.
I believe there was some push to make it 0.7 instead of 2.0 ....

closed 2024-09-22 by wiedehopf