wiedehopf/adsb-scripts #40: up fr24feed version & Setup instructions

afe1520860, 2024-05-22:
`- wget -O fr24.deb https://repo-feed.flightradar24.com/rpi_binaries/fr24feed_1.0.46-2_armhf.deb`
`+ wget -O fr24.deb https://repo-feed.flightradar24.com/rpi_binaries/fr24feed_1.0.48-0_armhf.deb`

update manual:

```
Step 1.3 - Would you like to participate in MLAT calculations? (yes/no)$:
no

Step 3.A - Enter antenna's latitude (DD.DDDD)
DD.DDDD

Step 3.B - Enter antenna's longitude (DDD.DDDD)
DDD.DDDD

Step 3.C - Enter antenna's altitude above the sea level (in feet)
HHH

Step 4.1 - Receiver selection:
Enter your receiver type (1-7)$:
4 (ModeS Beast ... yes really, no it doesn't matter what receiver you're actually running, this means a beast protocol connection to readsb, the decoder at the top of this guide)

Step 4.2 - Please select connection type:
Enter your connection type (1-2)$:
1 (Network ... yes really, it will connect to the decoder at the top of this guide using local loopback)

Step 4.3A - Please enter your receiver's IP address/hostname (127.0.0.1 is correct for everyone, means same computer)
127.0.0.1

Step 4.3B - Please enter your receiver's data port number
30005

Step 5.1 - Would you like to enable RAW data feed on port 30334 (yes/no)$:
no

Step 5.2 - Would you like to enable Basestation data feed on port 30003 (yes/no)$:
no
```

wiedehopf, 2024-05-22:
Not sure what you're asking for.

And no i don't generally up the software version so soon after release, fr24 stuff is always buggy.

afe1520860, 2024-05-24:
Update [fr24-nopackage.sh](https://github.com/wiedehopf/adsb-scripts/blob/master/fr24-nopackage.sh) because FR24 changed the version

in addition FR24 changed the setup questions, thats a part to change https://github.com/wiedehopf/adsb-wiki/wiki/Raspbian-Lite:-ADS-B-receiver#7-other-feed-clients-optional

closed 2024-05-24 by wiedehopf