wiedehopf/adsb-scripts #41: Install script fails at line 62 in a LXC container

mihanson, 2024-07-03:
I was trying to install readsb via your install script and the script throws a "Permission denied" error when trying to mount root as rw. The script is running under sudo and I am inside a LXC container.

```
mihanson@ads:~$ sudo bash -c "$(wget -O - https://github.com/wiedehopf/adsb-scripts/raw/master/readsb-install.sh)"
--2024-07-02 19:35:28--  https://github.com/wiedehopf/adsb-scripts/raw/master/readsb-install.sh
Resolving github.com (github.com)... 140.82.116.4
Connecting to github.com (github.com)|140.82.116.4|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://raw.githubusercontent.com/wiedehopf/adsb-scripts/master/readsb-install.sh [following]
--2024-07-02 19:35:28--  https://raw.githubusercontent.com/wiedehopf/adsb-scripts/master/readsb-install.sh
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 185.199.109.133, 185.199.111.133, 185.199.110.133, ...
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|185.199.109.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 10590 (10K) [text/plain]
Saving to: 'STDOUT'
                                                                     100%[===================================================================================================================================================================================>]  10.34K  --.-KB/s    in 0s      

2024-07-02 19:35:29 (161 MB/s) - written to stdout [10590/10590]

[sudo] password for mihanson: 
665325 (process ID) old priority 0, new priority 10
mount: /: permission denied.
       dmesg(1) may have more information after failed mount system call.
[ERROR] Error in line 62 when executing: mount -o remount,rw /
```

```
mihanson@ads:~$ mount
/dev/rbd0 on / type ext4 (rw,relatime,stripe=16)
none on /dev type tmpfs (rw,relatime,size=492k,mode=755,uid=100000,gid=100000,inode64)
proc on /proc type proc (rw,nosuid,nodev,noexec,relatime)
proc on /proc/sys/net type proc (rw,nosuid,nodev,noexec,relatime)
proc on /proc/sys type proc (ro,relatime)
proc on /proc/sysrq-trigger type proc (ro,relatime)
sysfs on /sys type sysfs (ro,nosuid,nodev,noexec,relatime)
sysfs on /sys/devices/virtual/net type sysfs (rw,nosuid,nodev,noexec,relatime)
fusectl on /sys/fs/fuse/connections type fusectl (rw,nosuid,nodev,noexec,relatime)
udev on /dev/bus/usb/003 type devtmpfs (rw,nosuid,relatime,size=32565096k,nr_inodes=8141274,mode=755,inode64)
proc on /dev/.lxc/proc type proc (rw,relatime)
sys on /dev/.lxc/sys type sysfs (rw,relatime)
none on /sys/fs/cgroup type cgroup2 (rw,nosuid,nodev,noexec,relatime)
lxcfs on /proc/cpuinfo type fuse.lxcfs (rw,nosuid,nodev,relatime,user_id=0,group_id=0,allow_other)
lxcfs on /proc/diskstats type fuse.lxcfs (rw,nosuid,nodev,relatime,user_id=0,group_id=0,allow_other)
lxcfs on /proc/loadavg type fuse.lxcfs (rw,nosuid,nodev,relatime,user_id=0,group_id=0,allow_other)
lxcfs on /proc/meminfo type fuse.lxcfs (rw,nosuid,nodev,relatime,user_id=0,group_id=0,allow_other)
lxcfs on /proc/slabinfo type fuse.lxcfs (rw,nosuid,nodev,relatime,user_id=0,group_id=0,allow_other)
lxcfs on /proc/stat type fuse.lxcfs (rw,nosuid,nodev,relatime,user_id=0,group_id=0,allow_other)
lxcfs on /proc/swaps type fuse.lxcfs (rw,nosuid,nodev,relatime,user_id=0,group_id=0,allow_other)
lxcfs on /proc/uptime type fuse.lxcfs (rw,nosuid,nodev,relatime,user_id=0,group_id=0,allow_other)
lxcfs on /sys/devices/system/cpu type fuse.lxcfs (rw,nosuid,nodev,relatime,user_id=0,group_id=0,allow_other)
udev on /dev/full type devtmpfs (rw,nosuid,relatime,size=32565096k,nr_inodes=8141274,mode=755,inode64)
udev on /dev/null type devtmpfs (rw,nosuid,relatime,size=32565096k,nr_inodes=8141274,mode=755,inode64)
udev on /dev/random type devtmpfs (rw,nosuid,relatime,size=32565096k,nr_inodes=8141274,mode=755,inode64)
udev on /dev/tty type devtmpfs (rw,nosuid,relatime,size=32565096k,nr_inodes=8141274,mode=755,inode64)
udev on /dev/urandom type devtmpfs (rw,nosuid,relatime,size=32565096k,nr_inodes=8141274,mode=755,inode64)
udev on /dev/zero type devtmpfs (rw,nosuid,relatime,size=32565096k,nr_inodes=8141274,mode=755,inode64)
devpts on /dev/pts type devpts (rw,nosuid,noexec,relatime,gid=100005,mode=620,ptmxmode=666,max=1026)
devpts on /dev/ptmx type devpts (rw,nosuid,noexec,relatime,gid=100005,mode=620,ptmxmode=666,max=1026)
devpts on /dev/console type devpts (rw,nosuid,noexec,relatime,gid=100005,mode=620,ptmxmode=666,max=1026)
devpts on /dev/tty1 type devpts (rw,nosuid,noexec,relatime,gid=100005,mode=620,ptmxmode=666,max=1026)
devpts on /dev/tty2 type devpts (rw,nosuid,noexec,relatime,gid=100005,mode=620,ptmxmode=666,max=1026)
none on /proc/sys/kernel/random/boot_id type tmpfs (ro,nosuid,nodev,noexec,relatime,size=492k,mode=755,uid=100000,gid=100000,inode64)
tmpfs on /dev/shm type tmpfs (rw,nosuid,nodev,uid=100000,gid=100000,inode64)
tmpfs on /run type tmpfs (rw,nosuid,nodev,size=13039580k,nr_inodes=819200,mode=755,uid=100000,gid=100000,inode64)
tmpfs on /run/lock type tmpfs (rw,nosuid,nodev,noexec,relatime,size=5120k,uid=100000,gid=100000,inode64)
mqueue on /dev/mqueue type mqueue (rw,nosuid,nodev,noexec,relatime)
ramfs on /run/credentials/systemd-sysctl.service type ramfs (ro,nosuid,nodev,noexec,relatime,mode=700)
ramfs on /run/credentials/systemd-sysusers.service type ramfs (ro,nosuid,nodev,noexec,relatime,mode=700)
ramfs on /run/credentials/systemd-tmpfiles-setup-dev.service type ramfs (ro,nosuid,nodev,noexec,relatime,mode=700)
ramfs on /run/credentials/systemd-tmpfiles-setup.service type ramfs (ro,nosuid,nodev,noexec,relatime,mode=700)
tmpfs on /var/cache/fontconfig type tmpfs (rw,nosuid,nodev,relatime,size=32000k,mode=755,uid=100000,gid=100000,inode64)
tmpfs on /run/user/1000 type tmpfs (rw,nosuid,nodev,relatime,size=6519788k,nr_inodes=1629947,mode=700,uid=101000,gid=101000,inode64)
```

closed 2024-07-03 by wiedehopf

wiedehopf, 2024-07-03:
So is it working now?

mihanson, 2024-07-03:
Yes! Thank you!
```
Jul 03 15:41:14 ads piaware[1111703]: piaware received a message from readsb!
Jul 03 15:41:14 ads piaware[1111703]: piaware has successfully sent several msgs to FlightAware!
```