wiedehopf/adsb-scripts #42: readsb service didn't start

Maksym68, 2024-09-06:
Sep 06 08:20:30 pi24-bookworm readsb[2749]: [2024-09-06 08:20:30.088 UTC] readsb starting up.
Sep 06 08:20:30 pi24-bookworm readsb[2749]: readsb version: 3.14.1639 wiedehopf git: 123fbf1 (committed: Sun Aug 25 20:16:54 2024 0200)
Sep 06 08:20:30 pi24-bookworm readsb[2749]: 30002: Raw TCP output port
Sep 06 08:20:30 pi24-bookworm readsb[2749]: 30005: Beast TCP output port
Sep 06 08:20:30 pi24-bookworm readsb[2749]: 30003: SBS TCP output ALL port
Sep 06 08:20:30 pi24-bookworm readsb[2749]: 30001: Raw TCP input port
Sep 06 08:20:30 pi24-bookworm readsb[2749]: 30004: Beast TCP input port
Sep 06 08:20:30 pi24-bookworm readsb[2749]: 30104: Beast TCP input port
Sep 06 08:20:30 pi24-bookworm readsb[2749]: FATAL: rtlsdr: no supported devices found.
Sep 06 08:20:30 pi24-bookworm readsb[2749]: [2024-09-06 08:20:30.108 UTC] sdrOpen() failed, exiting!
Sep 06 08:20:30 pi24-bookworm readsb[2749]: [2024-09-06 08:20:30.115 UTC] Abnormal exit. uptime:  0d  0h  0m 0.032s
Sep 06 08:20:30 pi24-bookworm systemd[1]: readsb.service: Main process exited, code=exited, status=1/FAILURE
Sep 06 08:20:30 pi24-bookworm systemd[1]: readsb.service: Failed with result 'exit-code'.
Sep 06 08:20:45 pi24-bookworm systemd[1]: readsb.service: Scheduled restart job, restart counter is at 1.
Sep 06 08:20:45 pi24-bookworm systemd[1]: Stopped readsb.service - readsb ADS-B receiver.
Sep 06 08:20:45 pi24-bookworm systemd[1]: Started readsb.service - readsb ADS-B receiver.
Sep 06 08:20:45 pi24-bookworm readsb[3077]: invoked by: /usr/bin/readsb --device 0 --device-type rtlsdr --gain -10 --ppm 0 --max-range 450 --write-json-every 1 --net --net-heartbeat 60 --net-ro-size 1250 --net-ro-interval 0.05 --net-ri-port 30001 --net-ro-port 30002 --net-sbs-port 30003 --net-bi-port 30004,30104 --net-bo-port 30005 --json-location-accuracy 2 --range-outline-hours 24 --write-json /run/readsb --quiet
Sep 06 08:20:45 pi24-bookworm readsb[3077]: [2024-09-06 08:20:45.277 UTC] readsb starting up.
Sep 06 08:20:45 pi24-bookworm readsb[3077]: readsb version: 3.14.1639 wiedehopf git: 123fbf1 (committed: Sun Aug 25 20:16:54 2024 0200)
Sep 06 08:20:45 pi24-bookworm readsb[3077]: 30002: Raw TCP output port
Sep 06 08:20:45 pi24-bookworm readsb[3077]: 30005: Beast TCP output port
Sep 06 08:20:45 pi24-bookworm readsb[3077]: 30003: SBS TCP output ALL port
Sep 06 08:20:45 pi24-bookworm readsb[3077]: 30001: Raw TCP input port
Sep 06 08:20:45 pi24-bookworm readsb[3077]: 30004: Beast TCP input port
Sep 06 08:20:45 pi24-bookworm readsb[3077]: 30104: Beast TCP input port
Sep 06 08:20:45 pi24-bookworm readsb[3077]: FATAL: rtlsdr: no supported devices found.
Sep 06 08:20:45 pi24-bookworm readsb[3077]: [2024-09-06 08:20:45.296 UTC] sdrOpen() failed, exiting!
Sep 06 08:20:45 pi24-bookworm readsb[3077]: [2024-09-06 08:20:45.303 UTC] Abnormal exit. uptime:  0d  0h  0m 0.026s
Sep 06 08:20:45 pi24-bookworm systemd[1]: readsb.service: Main process exited, code=exited, status=1/FAILURE
Sep 06 08:20:45 pi24-bookworm systemd[1]: readsb.service: Failed with result 'exit-code'.

wiedehopf, 2024-09-06:
```
Sep 06 08:20:45 pi24-bookworm readsb[3077]: FATAL: rtlsdr: no supported devices found.

```

Maksym68, 2024-09-06:
pi@pi24-bookworm:~ $ rtl_eeprom
Found 1 device(s):
  0:  Generic RTL2832U OEM

Using device 0: Generic RTL2832U OEM
Detached kernel driver
Found Rafael Micro R820T tuner

Current configuration:
__________________________________________
Vendor ID:              0x0bda
Product ID:             0x2838
Manufacturer:           Nooelec
Product:                UAT
Serial number:          00001090
Serial number enabled:  yes
Bias Tee always on:     no
Remote wakeup enabled:  no

Maksym68, 2024-09-06:
Sep 06 11:21:26 pi24-bookworm readsb[33342]: readsb version: 3.14.1639 wiedehopf git: 123fbf1 (committed: Sun Aug 25 20:16:54 2024 0200)
Sep 06 11:21:26 pi24-bookworm readsb[33342]: Using lat: 51.9100, lon: 20.7200
Sep 06 11:21:26 pi24-bookworm readsb[33342]: 30002: Raw TCP output port
Sep 06 11:21:26 pi24-bookworm readsb[33342]: 30005: Beast TCP output port
Sep 06 11:21:26 pi24-bookworm readsb[33342]: 30003: SBS TCP output ALL port
Sep 06 11:21:26 pi24-bookworm readsb[33342]: 30001: Raw TCP input port
Sep 06 11:21:26 pi24-bookworm readsb[33342]: 30004: Beast TCP input port
Sep 06 11:21:26 pi24-bookworm readsb[33342]: 30104: Beast TCP input port
Sep 06 11:21:26 pi24-bookworm readsb[33342]: FATAL: rtlsdr: no supported devices found.
Sep 06 11:21:26 pi24-bookworm readsb[33342]: [2024-09-06 11:21:26.545 UTC] sdrOpen() failed, exiting!
Sep 06 11:21:26 pi24-bookworm readsb[33342]: [2024-09-06 11:21:26.553 UTC] Abnormal exit. uptime:  0d  0h  0m 0.031s
Sep 06 11:21:26 pi24-bookworm systemd[1]: readsb.service: Main process exited, code=exited, status=1/FAILURE
Sep 06 11:21:26 pi24-bookworm systemd[1]: readsb.service: Failed with result 'exit-code'.
Sep 06 11:21:28 pi24-bookworm systemd[1]: Stopped readsb.service - readsb ADS-B receiver.
Sep 06 11:21:28 pi24-bookworm systemd[1]: Started readsb.service - readsb ADS-B receiver.
Sep 06 11:21:28 pi24-bookworm readsb[33407]: invoked by: /usr/bin/readsb --device 0 --device-type rtlsdr --gain 43.9 --ppm 0 --lat 51.9100 --lon 20.7200 --max-range 450 --write-json-every 1 --net --net-heartbeat 60 --net-ro-size 1250 --net-ro-interval 0.05 --net-ri-port 30001 --net-ro-port 30002 --net-sbs-port 30003 --net-bi-port 30004,30104 --net-bo-port 30005 --json-location-accuracy 2 --range-outline-hours 24 --write-json /run/readsb --quiet
Sep 06 11:21:28 pi24-bookworm readsb[33407]: [2024-09-06 11:21:28.760 UTC] readsb starting up.
Sep 06 11:21:28 pi24-bookworm readsb[33407]: readsb version: 3.14.1639 wiedehopf git: 123fbf1 (committed: Sun Aug 25 20:16:54 2024 0200)
Sep 06 11:21:28 pi24-bookworm readsb[33407]: Using lat: 51.9100, lon: 20.7200
Sep 06 11:21:28 pi24-bookworm readsb[33407]: 30002: Raw TCP output port
Sep 06 11:21:28 pi24-bookworm readsb[33407]: 30005: Beast TCP output port
Sep 06 11:21:28 pi24-bookworm readsb[33407]: 30003: SBS TCP output ALL port
Sep 06 11:21:28 pi24-bookworm readsb[33407]: 30001: Raw TCP input port
Sep 06 11:21:28 pi24-bookworm readsb[33407]: 30004: Beast TCP input port
Sep 06 11:21:28 pi24-bookworm readsb[33407]: 30104: Beast TCP input port
Sep 06 11:21:28 pi24-bookworm readsb[33407]: FATAL: rtlsdr: no supported devices found.
Sep 06 11:21:28 pi24-bookworm readsb[33407]: [2024-09-06 11:21:28.785 UTC] sdrOpen() failed, exiting!
Sep 06 11:21:28 pi24-bookworm readsb[33407]: [2024-09-06 11:21:28.792 UTC] Abnormal exit. uptime:  0d  0h  0m 0.032s
Sep 06 11:21:28 pi24-bookworm systemd[1]: readsb.service: Main process exited, code=exited, status=1/FAILURE
Sep 06 11:21:28 pi24-bookworm systemd[1]: readsb.service: Failed with result 'exit-code'.

Maksym68, 2024-09-06:
pi@pi24-bookworm:~ $ rtl_eeprom
Found 1 device(s):
  0:  Generic RTL2832U OEM

Using device 0: Generic RTL2832U OEM
Detached kernel driver
Found Rafael Micro R820T tuner

Current configuration:
__________________________________________
Vendor ID:              0x0bda
Product ID:             0x2838
Manufacturer:           Nooelec
Product:                NESDR SMArt v5
Serial number:          1090
Serial number enabled:  yes
Bias Tee always on:     no
Remote wakeup enabled:  no
__________________________________________
Reattached kernel driver

wiedehopf, 2024-09-06:
Haven't seen that before.
Try a reboot or rerunning the install script: <https://github.com/wiedehopf/adsb-scripts/wiki/Automatic-installation-for-readsb>

closed 2024-09-22 by wiedehopf