wiedehopf/adsb-scripts #45: Changing gain - improved check for strog messages

pilaGit, 2025-08-14:
Currently several versions of command to get get the percentage of strong messages are seen. They could be improved to work without any problems on any system and be much simpler by using only the awk. 

a) Nice and tidy version, if user knows which file exists - just edit the filename at the end as needed: 

`awk -F '[\\[\\]\":,]' '/\"total/ { printf "\nPercentage of strong messages: %.3f\nSignal: %.1f\nNoise: %.1f\n", $65 * 100 / ($47+$48), $53, $57 } ' < "/run/readsb/stats.json"
`

b) Big ugly version that works for all without knowing anything, the same way as the original one:

`cat /run/{dump1090,dump1090-fa,readsb}/stats.json 2>/dev/null | awk -F '[\\[\\]\":,]' '/\"total/ { printf "\nPercentage of strong messages: %.3f\nSignal: %.1f\nNoise: %.1f\n", $65 * 100 / ($47+$48), $53, $57 }'`

Hope this may be useful touch in your great work! Many thanks for it!

pilaGit, 2025-08-14:
I have seen different versions on the page about adjusting gain and various versions in many posts over last years. 

I apologize If the issue is obsolete, my mistake. I am trying to setup my antenna so am reading about it over last two days trying to figure what can be done. 

Just ignore it, sorry to bother you.

closed 2025-08-14 by pilaGit

wiedehopf, 2025-08-14:
Oh are you proposing to use awk over jq?

I just assume people install jq when they get an error of it not being installed.

wiedehopf, 2025-08-14:
Hmm oh it already uses awk.
So i guess your command line is shorter.

Not sure if it relies on the structure of stats.json a bit too much though.

pilaGit, 2025-08-14:
It does rely on this exact number of fields being present. If fields are not fixed, I could simply rewrite it to use key words and read numbers after them. But, this was much shorter.

wiedehopf, 2025-08-14:
pi24 image, the horror :)

If you want to ever start with Raspberry Pi OS: https://github.com/wiedehopf/adsb-wiki/wiki/Raspbian-Lite:-ADS-B-receiver
That's a bit of a guide for bare metal.

But unless you like messing with the command line, just go with adsb.im and it's all configurable from the webinterface.

pilaGit, 2025-08-14:
For now, my system is running on bare metal RPi3. This is annoying since I have 50+ computers in the apartment. I am very stringent about power consumption and useless machines.  I will likely try to move it into a container under Libreelec which are used everywhere for TV/Video and run anyway on RPI4. 

Pi24 image was making me mad for two days until I finally understood enough to find your work :) Today I finally started to understand my problems wit the antenna I made yesterday. Thanks to you.

pilaGit, 2025-08-14:
I will check your above intel in few day, when I am satisfied how the system works. Thanks.